soligen2010/encoder #2: encoder tested with this library?

rocketbob, 2017-01-17:
I've tried two different encoders, one being an Alps and the other a Panasonic.  They both "sort of" work.  Right now the Panasonic works nearly correct in one direction, but when turning in the other it only bounces between two values.  Have tried various permutation of wiring and code changes.

Would like to know try encoders were tested with the library and go from there.

soligen2010, 2017-01-18:
I used the generic cheap ebay rotary encoders.  

In the .h file you can change the way the library interprets the encoder.  Perhaps you need different options.  Here is the part I am talking about

// ----------------------------------------------------------------------------

#define ENC_NORMAL        (1 << 1)   // use Peter Danneger's decoder
#define ENC_FLAKY         (1 << 2)   // use Table-based decoder

// ----------------------------------------------------------------------------

#ifndef ENC_DECODER
#  define ENC_DECODER     ENC_NORMAL   <<<<<<<<<<<<<<<<<<< You can change this
#endif

#if ENC_DECODER == ENC_FLAKY
#  ifndef ENC_HALFSTEP
#    define ENC_HALFSTEP  1      <<<<<<<<<<<<< You can change this when in FLAKY mode  // use table for half step per default
#  endif
#endif

In this default ENC_DECODER is set to ENC_NORMAL.  You can try to change it to ENC_FLAKY.  When you are using ENC_FLAKEY, you can set ENC_HALFSTEP to either 1 or 0

soligen2010, 2017-01-18:
Sorry about the formatting of the code.  I dont know why it formatted badly and my attempts to fix it failed.  Please look in the .h file

soligen2010, 2017-01-18:
Formatting is better.  Apparently git hub doesn't like stand along # characters in comments.  Indenting is removed, but I think you can get the idea

soligen2010, 2017-01-18:
For wiring, on the side with 2 pins, this is your button.  One pin to Ground, the other to your button input pin.

The side with 3 pins is the rotary encoder.  Center pin should be ground and the outer two should co to the two encoder pins.  If the encoder moves backwards from what is intended, just swap the pins in code.

soligen2010, 2017-02-13:
If you are still following this, I uploaded some changes that fixed an issue.  I don't know if it caused your issue of not, but you can give it a try.

Also be sure to set clicks per step properly.  The default was previously 1, but I changed the default to 4 as this seems to be very common on inexpensive encoders.

rocketbob, 2017-02-15:
Ok thank you will give it another go.  I ended up writing my own interrupt-based encoder routing which worked out to be much simpler and works great.  On the Due since every IO pin is interrupt-able its easy to do.

void encoderPinInterruptHandler() {
	int msb = digitalRead(ENCODER_A);
	int lsb = digitalRead(ENCODER_B);
	int currentEncoderValue = (msb << 1) | lsb;
	int sum = (lastEncoderValue << 2) | currentEncoderValue;
	if (sum == 0b0100 || sum == 0b0010) encoderValue++;
	if (sum == 0b1110 || sum == 0b0111) encoderValue--;
	lastEncoderValue = currentEncoderValue;
}

soligen2010, 2017-02-24:
Haven't heard any back, so closing issue

closed 2017-02-24 by soligen2010