soligen2010/encoder #3: Encoder only works with #define WITHOUT_BUTTON 1

evetek, 2017-01-31:
Hey,

If I run the example, the values don't change at all and button presses are not detected.

The strange thing is that everything works perfect if I add #define WITHOUT_BUTTON 1 to the start.
Obviously the button does not work then, whats going on? how can I get the button to work also.

Thanks.

soligen2010, 2017-02-01:
Could you please try this version from before I added support for multiplexed analog buttons:
[https://github.com/soligen2010/encoder/tree/b1ee27635bb168f6484eb1c44e99731ee4d79e9d](https://github.com/soligen2010/encoder/tree/b1ee27635bb168f6484eb1c44e99731ee4d79e9d)

I currently don't have an encoder circuit to test with. They worked together previously, so if you can help me figure out which commit broke it, I would appreciate it.

eversins, 2017-02-01:
Thank you for your support, your posted version worked flawlessly. It even fixed the problems I had with 0xPIT version.

Thank you so much.

soligen2010, 2017-02-01:
Coo, I am glad it worked.  Can you do me a favor and also try this version:

[https://github.com/soligen2010/encoder/tree/9efede6b90c3588dab8939aed1657d991bb4a962](https://github.com/soligen2010/encoder/tree/9efede6b90c3588dab8939aed1657d991bb4a962)

This will help me narrow down where the problem was introduced.

Thanks

eversins, 2017-02-01:
https://github.com/soligen2010/encoder/tree/9efede6b90c3588dab8939aed1657d991bb4a962

Does not work.

soligen2010, 2017-02-01:
> 991bb4a962

OK thanks. I don't see anything obvious wrong, but at least that narrows it
down for me

soligen2010, 2017-02-01:
I think I found the issue.  Can you try again with the most current version of the library (the one that doesn't work) and change the call to the constructor to pass the 4th parameter instead of letting it default (the default value is 1)

Change
     encoder = new ClickEncoder(A1, A0, A2);
to
     encoder = new ClickEncoder(A1, A0, A2,1);

I think the compiler is choosing the wrong constructor and specifying the 4th parameter should clear the ambiguity.

If you could, please try this and let me know.

Thanks

eversins, 2017-02-02:
> encoder = new ClickEncoder(A1, A0, A2,1);

fixed it.
Thank you.

soligen2010, 2017-02-13:
I corrected the original problem with the latest commit.  The 3 parameter constructor as in the example now works, however I would still recommend specifying the 4th parameter (stepsPerNotch) as that can vary by brand of the encoder.

BTW, I change the default steps per notch from 1 to 4 as this seems to be what the inexpensive encoders use.

I am going to close this issue, please open another issue if there is a problem.

closed 2017-02-13 by soligen2010