soligen2010/encoder #5: request

robertodevet, 2017-06-15:
Hello , i am using your click encoder program and find it the fastest i tested lately.
In the serial i get my values and button chances ,so that is perfect.
But ,due to my lack of skills , i can't figure out how i can implement it in my sketch.
I only want to use if encodervalue increases to get "right = TRUE " or UP and when the value decreases  a "left = TRUE".
Can you please give me a hint how to implement this , i want to use it to steer an steppenmotor.

soligen2010, 2017-06-15:
You would store the last value you got from the encoder, then compare the new value from the encoder to the last value to see if it went up or down.

robertodevet, 2017-06-15:
Thanks for the reply.
I have being playing around with a lastValue and compare with the value but cant get it working as i want yet.
In your clickencoder.h you have include a getDirection , can't i use this for my sketch to determine an UP or Down ?

soligen2010, 2017-06-15:
There is not a getDirection in clickencoder.h.  Perhaps you are looking at a different library.  I don't think a getDirection method would fit very well with how this library is constructed.  I think using a last value approach is the best bet, but the nice thing about open source is that you are free to copy this library and add any feature that you like.  You may also consider not doing the direction thing at all and just using the value returned  where 0 is straight and the farther away from zero you are positive or negative is how far you should be turned left or right.  This is the way that I would do it rather than what you are trying to do.

robertodevet, 2017-06-16:
I think you are right ,that i have an adjusted library.
Will look into your suggestion , but as a novice it's kind of hard ,but i will get there .

soligen2010, 2017-06-19:
I hope it all works out for you.  I am going to close this issue, if you later have an issue with the library, please open another issue.

closed 2017-06-19 by soligen2010