soligen2010/encoder #6: Button clicked response time

R5fan, 2017-10-13:
Hi, just testing your library on a ESP32 using your ESP8266Example.
The encoder itself seems to work well, and responsive. However it takes a fairly long time before I get a "Button: ClickEncoder::Clicked" message. Im guessing on the order of ~500ms or so, too long for responsive menu's. I tried changing ENC_BUTTONINTERVAL , but that seems to make no difference?

Also, Im a little confused about the difference between ClickEncoder::Pressed and ClickEncoder::Clicked. The former never seems to trigger?

thanks!

soligen2010, 2017-10-13:
There is a delay after click to see if a double click happens.  If  you don't use double clicks, then disable doubleclicks and clicks should be very responsive.  If you need both click and doubleclick events, then there isn't really a way get rid of the delay, however you can change the double click timing to find the happy medium you prefer. For something fast like I menu navigation, I suggest designing your UI to avoid needing double clicks.  The example sketch has everything enabled for example purposes.

ClickEncoder::Pressed has never been implemented, so you can ignore it (same with closed).  It was in the original library I forked, and I don't know the intent behind these were, so next time I am updating the code I think I'll just remove them

R5fan, 2017-10-13:
Thanks, Ill try that, that makes sense.
One more question if I may; for my application, I want to change some settings when the button is pressed down while being turned. Any suggestion on how to do that best?

septillion-git, 2017-10-13:
I have something similar in the pipe line. That's why I would like all state to be implemented including pressed and clicked :). So please don't remove them, include them :D

soligen2010, 2017-10-16:
R5Fan, you can use the Held and Released events.  If it is held for a breif period you get held/released instead of clicked

Septillion.  Sure, I'll leave them in for now. I am not sure how Pressed would differ from Held, but I would be happy to see a pull request for added functionality :)

R5fan, 2017-10-16:
If Im not mistaken, Held event only triggers after BTN_HOLDTIME. For me thats not a big issue, might even be desirable, but in other applications one might want to "push and turn" with instant response.

soligen2010, 2017-10-16:
Yes with the current implementation you would have to choose a balance between the button hold time and how fast you have to be to get a click event.  If we implement a pressed event, then I think it may make more sense to make it a separate method instead if using the button enum.  Just an independent method to return this state.  Otherwise it would get confused with the currently implemented button states because multiple states could occur at the same time.

I think with the current implementation push and turn with instant response is do-able as long as you don't also want to process click events from the same button.

soligen2010, 2017-10-16:
I think an independent pressed method would be pretty easy to implement.  A boolean method that returns false if keyDownTicks == 0 otherwise return true.

Do one of you guys want to give it a try and see of this works properly?

R5fan, 2017-10-16:
IM happy to test, but Im very much an Arduino noob, and still struggling even with the example.. Specifically, I simply dont understand how this code works:

  #define VERBOSECASE(label) case label: Serial.println(#label); break;
    switch (b) {
      VERBOSECASE(ClickEncoder::Pressed);
      VERBOSECASE(ClickEncoder::Held)
      VERBOSECASE(ClickEncoder::Released)
      VERBOSECASE(ClickEncoder::Clicked)

I can copy paste and make it work, but I just dont understand it. Care to explain a little?

soligen2010, 2017-10-16:
VERBOSECASE is a macro.  Whatever is in the parens is plugged in for label.

so this:  VERBOSECASE(ClickEncoder::Pressed)
becomes this: caseClickEncoder:Pressed: Serial.println(#ClickEncoder::Pressed); break;

Its just a way to not have to type the same thing over and over again.

All this is straight C++ (not Arduino specific) so I recommend you find a tutorial on C++ to learn the basics.  After knowing the basics, you can google C++ SWITCH or C++ #DEFINE and usually fins some good resources and examples.

If it is the switch(b) statement you don't understand then definitely find a C++ programming course or C++ reference site as that will explain it much better than I can here.

R5fan, 2017-10-16:
Its the macro concept thats new to me, and now I know what to google for, I will, thanks a lot.

septillion-git, 2017-10-16:
Mm, yeah, doing it via a separate method would be easy but would in my opinion break the nice enum structure.

I would implement it in the enum return but a separate method as well. So you go from pressed to held to released or pressed, clicked etc. So you can make a distinction between long press, click, press and rotate and long press and rotate etc. :)

But I have a bit of a back log 👼

soligen2010, 2017-10-16:
Seems overloaded to me to include both the event (clicked, etc) and current button state (pressed or not) in one data element.  The data architect in me shivers a bit at overloading a data element like this, but I am open to anything that turns out to work well.

soligen2010, 2018-03-05:
Hi guys,  Just wanted to check and see if anyone is working up changes for this issue, or if I should just close it.  Please post an update.

tr4nt0r, 2018-07-13:
may i suggest, that instead of waiting if the second click  occurs, you simply return the click event as soon as it is recognized and when the last click was less than "BTN_DOUBLECLICKTIME" ago, return a doubleclick event. This would reduce the response time for the click event and there shouldn't be any problems with double click either. The only case this could break functionality would be if someone has completely different actions for click and doubleclick. You could use a define to enable instant response and by default it is disabled

soligen2010, 2018-07-26:
You can speed up click events by disabling the double click events.  For me, I have often implemented 2 completely independent responses for click vs. double click where it would be incorrect to issue the click event along with the double click.  Issuing both events would make it inconvenient for the main code to distinguish these different cases when it needed it to.  I am going to choose to not implement your suggestion in this fork, but the beauty of open source is that you can fork this library and customize this to work however is best for you :)

closed 2021-01-23 by soligen2010