soligen2010/encoder #7: request: timer interrupt example for ESP

R5fan, 2017-10-23:
The example for the esp doesnt use a timer interrupt because of the potential interference with wifi; the AVR example does, but doesnt compile on ESP because the timer library seems incompatible.

Thing is, I dont need wifi, but I do need a timer interrupt because the graphics library Im using is blocking. If anyone could provide me with an example how to use encoder.h with a timer interrupt on ESP, I would be most grateful.

soligen2010, 2017-10-23:
Googleing turned up a timer example here:

https://hackaday.io/project/12818-not-grbl/log/47580-esp8266-timer0-and-isr

You will have to take this example and combine it with encoder to get something to work. When you get an example working, please make a pull request to contribute it.

The simpler answer may be to choose a different MCU if you don't need wifi.  A lot of people experience stability issues with the ESP chip, so it may not be the best choice for you.

R5fan, 2017-10-23:
I just got it working, based on this:
https://techtutorialsx.com/2017/10/07/esp32-arduino-timer-interrupts/

Just insert encoder.service();  in the onTimer function, increase the interrupt frequency to 1000 and remove the unnecessary bits. Works like a charm! I'll see if I can clean up my code and how to make a pull request.

As for a different MCU; Im loving this thing! Im not using wifi or bluetooth atm, but its nice to have the option in the future and it could be real useful. And I need a lot of IO and ram; driving a microstepper controller, eink display with no framebuffer,  encoder, buttons, RS232 and in the future likely some pressure sensors. My first prototype used 2 arduino nano's  I linked via SPI and that still wasnt enough.

If you're curious what Ill be using it for, and your library for; its a glider variometer. This is the current prototype:
https://i.imgur.com/oyn4zX5.jpg
This is an older one "in action" (using 2 nano's):
https://www.youtube.com/watch?v=7Zbjo2WrcWk

soligen2010, 2017-10-23:
Nice project.  Are you going to open source the code?  I am assuming this is for real gliders (not RC glider).

If you are interested in RC stuff, you might be interested in my project 

https://github.com/soligen2010/RC_RX_CABELL_V3_FHSS

Although with the radio hardware I am using I am not sure I get enough range for gliders.

I see you are using the ESP32, which I haven't used (i'll need to look at it more closely).  I had assumed you meant the ESP8266.  The example was for the 8266, and the timer concern was for that chip.  I don't know if the same timer concern applies to the ESP32.  I would be quite happy to have an ESP32 example added.

R5fan, 2017-10-23:
Here is an example using timer interrupt on ESP32 (should work on ESP8266 too I think). 
`
// code adopted from https://techtutorialsx.com/2017/10/07/esp32-arduino-timer-interrupts/
#include <ClickEncoder.h>

#define ENCODER_PINA     33
#define ENCODER_PINB     27
#define ENCODER_BTN      12
#define ENCODER_STEPS_PER_NOTCH    4  

ClickEncoder encoder = ClickEncoder(ENCODER_PINA,ENCODER_PINB,ENCODER_BTN,ENCODER_STEPS_PER_NOTCH);
int16_t last, value;
volatile int interruptCounter;
int totalInterruptCounter;
 
hw_timer_t * timer = NULL;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;

void onTimer() {
  portENTER_CRITICAL_ISR(&timerMux);
  interruptCounter++;
  portEXIT_CRITICAL_ISR(&timerMux);
  encoder.service();  
  value += encoder.getValue();  
  if (value != last) {
      last = value;
      Serial.print("Current encoder value");// serial writes are generally not a good idea in an interrupt. But this is only executed when the encoder is used.
      Serial.println(value);  
  } 
}

void setup() {
  Serial.begin(19200);  
  timer = timerBegin(0, 80, true);
  timerAttachInterrupt(timer, &onTimer, true);
  timerAlarmWrite(timer, 1000, true); // check encoder every 1000 microseconds
  timerAlarmEnable(timer);
  
  encoder.setButtonHeldEnabled(false);
  encoder.setDoubleClickEnabled(false);
  encoder.setButtonOnPinZeroEnabled(true);    
}

void loop() {
 // do other stuff
  Serial.println("Doing some long blocking stuff");
  delay (5000);  
}
`
And yes, this is for a real glider, to be used in combination with openvario, an opensource hard/software project for a glide computer. And I will opensource everything, the 3D designs are already on thingiverse. The code is too, ahm.. immature for publication.

But I fly RC too -and I have a taranis, and a multimodule. I will check out your other project :)

soligen2010, 2018-01-09:
This issue seems resolved.  If there is a reason to leave it open, please let me know or else I will close it.

Thanks

closed 2018-01-15 by soligen2010