soligen2010/encoder #10: Does not pick up the encoder truning

JacoFourie, 2018-01-14:
Hi. Thanks for fixing the lib. I use the original a lot on Arduino but now are starting with ESP8266 and 32. 
I have hooked up the encoder. It does pick up the click but does not see the encode values/ I can see a led flash on the ESP8266 when I turn the encoder but the value stays 0.
The click is also missed sometimes. 

Here is the code. What am I doing wrong ?

`
#include <ArduinoJson.h>
#include <ESP8266WiFi.h>
#include <PubSubClient.h>
#include <RunningMedian.h>
#include <SHT1x.h>
#include<CountUpDownTimer.h>
#include <Adafruit_SSD1306.h>
#include <ClickEncoder.h>

#define ENCODER_PINA     16  //D0 
#define ENCODER_PINB     12  //D6
#define ENCODER_BTN      0   //D3

#define ENCODER_STEPS_PER_NOTCH    4   // Change this depending on which encoder is used

ClickEncoder encoder = ClickEncoder(ENCODER_PINA,ENCODER_PINB,ENCODER_BTN,ENCODER_STEPS_PER_NOTCH);

static int16_t last, value;

#define OLED_RESET LED_BUILTIN //4
Adafruit_SSD1306 display(OLED_RESET);

// Specify data and clock connections and instantiate SHT1x object
#define dataPin  2
#define clockPin 14
SHT1x sht1x(dataPin, clockPin);

float temp_c , wbt; 
float humidity;
char payload[8];
String send_data;
char attributes[100];

CountUpDownTimer sensor_timer(UP, HIGH); 
CountUpDownTimer send_data_timer(UP, HIGH); 

#define TOKEN "123"

const char* ssid = "Pandora-LR";
const char* password = "123";
const char* mqtt_server = "demo.thingsboard.io";
const char* esp_hostname = "esptest";

// Static IP's
IPAddress arduino_ip ( 10,  64,   9,  229);
IPAddress dns_ip     ( 10,  64,   9,   20);
IPAddress gateway_ip ( 10,  64,   9,   20);
IPAddress subnet_mask(255, 255, 255,   0);

WiFiClient espClient;
PubSubClient client(espClient);

long lastMsg = 0;
char msg[50];

const char* outTopic = "espout";
const char* inTopic = "espin";

//d7 (13) en d8 (15) is ok op reboot.

int feeder_relay_pin = 15;
int stoker_relay_pin = 13;
bool relayState = LOW;
[... 282 lines cut ...]

// calc saturated vapour pressure at Celcius-temp T:
float CalcSVP(float T) {
  float tmp = aa * T / (bb+T);
  return svp_0 * pow(10,tmp);
}

// calc partial vapour pressure at Celcius-temp T, given wet-bulb Tw:
float CalcVapPress(float T,float Tw) {
  float A = 66.87 * (1+0.00115 * Tw); // at 1 atm!!
  return CalcSVP(Tw) - A * (T-Tw);
}

// calc relative humidity at Celcius-temp T, given wet-bulb Tw:
float CalcRH(float T,float Tw) {
  float tmp = (CalcVapPress(T,Tw)/CalcSVP(T)*1000+0.5)/10;
  if (tmp<0) {   
    return 0;
  }
  else {
    return tmp;
  }
}

`

soligen2010, 2018-01-15:
Please try the ESP8266 example and make sure it works.  If it works, then
it is likely something else in your code messing things up, or too slow of
a loop time. In that case you should start stripping everything out of your
sketch to get the minimum amount of code that exhibits the problem, or see
at what point the problem goes away. A possibility is that something you
are doing is just too slow so things get missed, in which case identifying
it may help you figure out what to speed up. Could be something else too -
either way stripping it down should help find it.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

JacoFourie, 2018-01-15:
Hi. Thanks for the reply. 
I am using just the sample now and I found that if the ENCODER_STEPS_PER_NOTCH = 4 I will get only one 

'''
Encoder Value: -1 
'''
and nothing after that. 

If I make it ENCODER_STEPS_PER_NOTCH = 1 I get this after each turn.

'''
Encoder Value: -2
Encoder Value: -1
'''
If I make it 2 it is the same as 4. On the Arduino I use 4 with this encode and then all works 100%.

soligen2010, 2018-01-16:
So all is good now? Different encoders need different steps per notch depending on how they are constructed.  Can the issue be closed?

soligen2010, 2018-01-16:
oops I accidentally clicked close

JacoFourie, 2018-01-16:
Nope it is not working. On the Arduino value will be 1 2 3 4 5 6 etc or -1 -2 -3 -4 -5  etc. depending on the direction I turn the encoder
Now I only get -1 -2 on each turn if the notch value is 1. Or it stays there or -1 and nothing more. I am using the same Encode as I use on the Arduino using the original lib. But now it does not work with 4. What else can I look at ?

JacoFourie, 2018-01-16:
So just to recap. If the notch value is 1 no mater the direction I turn one click it will spit out

Encoder Value: -2
Encoder Value: -1

So if I keep on turing the encoder I see this

Encoder Value: -2
Encoder Value: -1
Encoder Value: -2
Encoder Value: -1
Encoder Value: -2
Encoder Value: -1
Encoder Value: -2
Encoder Value: -1
Encoder Value: -2
Encoder Value: -1
Encoder Value: -2
Encoder Value: -1

soligen2010, 2018-01-17:
Nothing software related comes to mind, but of course that doesn't rule it out.  A possibility is that some pins on the ESP have pull up/down resistors.  Perhaps those are interfering with the inputs. Maybe check that and be sure you are using input pins that have no resistors or anything else connected to them.  This applies to both your own circuitry and what is on the ESP board you are using.  You can probe the input pins with an oscilloscope and see if the values are really changing.  You can try a multi meter, but the signal may change too fast for a multi meter to pick up.

JacoFourie, 2018-03-05:
I have this working on the ESP32 now. Using timer interrupts.

closed 2018-03-05 by JacoFourie