soligen2010/encoder #11: Each encoder indent results in multiple encoder events

moballa, 2018-01-14:
I have a rotatory encoder with pushbutton. I am using this library to read the values from the encoder. Each time I move the encoder one indent, 4 events get triggered advancing the counter by 4. Not sure if this is normal or my encoder is different than the usual expected behavior. I have tried adding #define ENC_DECODER (1 << 2) before including the .h file. I have also tried #define ENC_HALFSTEP with different values with 2 or 3 events getting triggered instead. When I set the 4th parameter of the constructor to 2, only 2 events get triggered but if I change it to the default of 1, 3 events get triggered.

Is there a problem with my encoder? Is there a way to ensure that only one event is triggered per encoder indent so that it only advances the position counter by 1 per indent?

Thanks for any help!

septillion-git, 2018-01-15:
Not a problem, jus a different type. It's the common China type which does a full step (4 states) for each detente. And you where close with the fourth parameter. If 2 halves the number of events, why didn't you try 4 ;) Aka, use 
```C++
encoder = new ClickEncoder(RotaryA, RotaryB, RotaryBtn, 4);
```

moballa, 2018-01-15:
I'm embarrassed - I tried so many other combinations but didn't try '4' in the forth parameter as per your suggestion.  Works perfectly now, thanks a bunch for responding!

septillion-git, 2018-01-16:
No problem!

soligen2010, 2018-01-16:
I'm glad it's working for you.  Closing issue

closed 2018-01-16 by soligen2010