soligen2010/encoder #12: Interrupt

GigaTech, 2018-02-01:
Hi Guys and Gals,

I am trying to 'squeeze' this onto a Arduino Nano (limited variable space) along with all the other code...

is there any way of calling just the rotary acceleration via an interrupt then the `<TimerOne.h>` would not be needed?
Encoder pinA = connected to = Arduino pin3 (being an interrupt pin)
Encoder pinB = connected to = Arduino pin4

```
const byte pinA = 3; // Pin on encoder, CLK
const byte pinB = 4; // Pin on encoder, DT
void setup(){
    // Encoder setup
    pinMode(pinA, INPUT);
    pinMode(pinB, INPUT);
    attachInterrupt(digitalPinToInterrupt(pinA), InterruptEncoder, LOW); // Attach the routine to service the interrupts
    ...
}

void InterruptEncoder(){ // Encoder interrupt
    ???
}
```

this code is currently working great but takes an age to wind to the top/bottom of the scale, enter acceleration...

Thanks

septillion-git, 2018-02-01:
There are tons of rotary encoder libraries that use external interrupts and just a few who poll. And you really want to change that single library to external interrupt as well?

Btw, Nano is as limited as a Uno. And if you don't wast memory with a book full of strings or worse, Strings, it has a lot of memory :)

soligen2010, 2018-02-02:
This is really a polling library so you can use this library without the timer interrupt simply by calling the service routine in your loop instead of using a timer interrupt.  The ESP6266 example does this, but you can do this on an arduino too. The only risk is that if your loop timing is too slow things may get missed, but I have used this technique with success, and then you don't need the timer library.  Alternately, you can set up the timer manually instead of using a library, which would have a very small code footprint compared to a library.  You can also add a 

#define  WITHOUT_BUTTON 1

this will remove code for handling buttons, which will also shrink the code footprint.

As for the acceleration, I think adding a new parameter to adjust the acceleration rate would be a good change.  If you work this out, please submit the enhancement.

soligen2010, 2018-03-05:
I am assuming this is resolved, so closing. Enter a new issue is needed

closed 2018-03-05 by soligen2010