soligen2010/encoder #13: ESP8266 example, micros() overflow

ocpBerlin, 2018-05-10:
Hi,

I've been using your fork in an Adafruit Huzzah ESP8266 with a rotary encoder and it has been working fine so far. But I noticed that the rotary encoder stops responding after a while...

Looking at the code, I think the problem is in this section from the loop function:

```
static uint32_t lastService = 0;

if (lastService + 1000 < micros()) {
       lastService = micros();
       encoder.service();
}
```

At some point, micros() will overflow (after 70 minutes or so) and then start over again. At that point lastService + 1000 is always going to be greater than micros(), so encoder.service() will not be called anymore.

I modified my code using the following:

```
 if (micros() < encoderLastService) {   // micros() has gone back to zero
    encoderLastService = micros();
  } else if (micros() > (encoderLastService + 1000)) { // Update only once every 1 ms  
    encoderLastService = micros();                
    encoder.service();  
  }
```

It seems to work for now!

Oscar.

septillion-git, 2018-05-10:
The classic "millis" rollover bug!

Don't do weird roll over checking. Completely unnecessary. Should just be
```C++
if (micros() - lastService >= 1000){
       lastService = micros();
       encoder.service();
}
```

soligen2010, 2021-01-23:
Fixed in latest commit

closed 2021-01-23 by soligen2010