soligen2010/encoder #14: Keyes encoder?

tablatronix, 2019-08-18:
I am having a heck of a time trying to get this to work properly, these are the keyes ebay modules pretty standard.

It seems to work but it actually does not work correctly. 

If I rotate CC I get value change on detent.
If I rotate CCW I get value change immediately! well before detent maybe half step. 

I have tried several settings and see no improvement, is it possible this code is not checking properly?

Ill try to scope it and see what is going on, tested several encoders I have.

Using esp8266 example

Ill make a video also to better explain, I am not sure I fully understand the flaky option and the half step modifiers, but I have tried them all.

tablatronix, 2019-08-18:
ok it looks like the library is not updating on the correct edge maybe

I am getting updates on first falling edge (ENC_A) on CCW

CW
[image]

CCW
[image]

CCW half turn(no detent)
[image]

tablatronix, 2019-08-18:
```C++
/*
 
This example demonstrates the use of the library on an ESP8266. without using a timer interrupt

 */

#include <ClickEncoder.h>
//#include <TimerOne.h>

#define ENCODER_PINA     4
#define ENCODER_PINB     5
#define ENCODER_BTN      -1

#define ENCODER_STEPS_PER_NOTCH    2   // Change this depending on which encoder is used

ClickEncoder encoder = ClickEncoder(ENCODER_PINA,ENCODER_PINB,ENCODER_BTN,ENCODER_STEPS_PER_NOTCH);

void setup() {
  Serial.begin(115200);

  // encoder.setButtonHeldEnabled(true);
  // encoder.setDoubleClickEnabled(true);
  
  // Enable the button to be on pin 0.  Normally pin 0 is not recognized as a valid pin for a button,
  // this is to maintain backward compatibility with an old version of the library
  // This version can have the button on pin zero, and this call enables the feature.
  // in this version best to use pin -1 instead of 0 to disable button functions
  encoder.setButtonOnPinZeroEnabled(true);
  pinMode(D3,OUTPUT);
}

void loop() { 

  //Call Service in loop becasue using timer interrupts may affect ESP8266 WIFI
  //however call no more than 1 time per millisecond to reduce encoder bounce
  static uint32_t lastService = 0;
  if (lastService + 1000 < micros()) {
    lastService = micros();                
    encoder.service();  
  }

 
static int16_t last, value;
    value += encoder.getValue();
  
  if (value != last) {
    digitalWrite(D3, LOW);
    Serial.print("Encoder Value: ");
    Serial.println(value);

    Serial.print(" Dir: ");
    Serial.print("\t");
    Serial.print(" Steps: ");
    Serial.print(value-last);
    Serial.print("\t");
    Serial.println((value > last) ? "10" : "20");
    last = value;
    digitalWrite(D3, HIGH);
  }
  
  ClickEncoder::Button b = encoder.getButton();
  if (b != ClickEncoder::Open) {
    Serial.print("Button: ");
    #define VERBOSECASE(label) case label: Serial.println(#label); break;
    switch (b) {
      VERBOSECASE(ClickEncoder::Pressed);
      VERBOSECASE(ClickEncoder::Held)
      VERBOSECASE(ClickEncoder::Released)
      VERBOSECASE(ClickEncoder::Clicked)
      VERBOSECASE(ClickEncoder::DoubleClicked)
    }
  }    
}

```

http://domoticx.com/arduino-rotary-encoder-keyes-ky-040/

tablatronix, 2019-08-18:
oops I should have been triggered on falling edge, still the same results though.

I am finding better luck using NUMSTEPS 1
and counting every 2 updates, seems to be working better

` if (abs(value - last) >= 2) {`

[image]

tablatronix, 2019-08-18:
ok it also looks like the 2 encoders I am using are different, one seems to be 4 steps, one is 2..
I thought they were the same..

Will keep testing, but this solution is working for me now, not sure how to get the lib to work better, Ill take a look at the actual logic one day and figure it out.

soligen2010, 2019-08-19:
I am glad you figured it out.  It would be nice if these things were more standard

tablatronix, 2019-08-19:
Yeah it is odd that I get proper function using steps 1 and counting 4 in my code, but if I use numsteps 4, the library does something different, might be a bug but I cannot say for sure, it is definitely updating a state early, I will look at the numsteps evaluation code as that is probably where a bug like this would be

tablatronix, 2019-08-20:
I think this is the problem.

Negative bit shifts.. When delta is -1
I am not sure how to correct atm, but `-1 >>= 1` (!=0)
So this does not work properly on CCW and fires early, but works fine on unsigned

```
  if (steps == 4) val >>= 2;
  if (steps == 2) val >>= 1; 
```

tablatronix, 2019-09-04:
Not sure what the most efficient way to do this but this is my solution.

```C++

  if (steps == 2) delta = val & 1;
  else if (steps == 4) delta = val & 3;
  else delta = 0; // default to 1 step per notch
 
  bool CCW = val < 0;
  if(CCW){
    delta *= -1;
    val = abs(val);
  }

  if (steps == 4) val >>= 2;
  if (steps == 2) val >>= 1;
  if(CCW) val *= -1;
```

tablatronix, 2020-01-05:
@soligen2010  thoughts?
Should I at least PR this so it can be reviewed or pulled for folks

soligen2010, 2020-01-05:
Yes send a PR.  Been a very long time since I looked at this, as as I am not the original author, this routine is a bit obscure.  What you are seems to make sense.  It seems like there should be a way to refactor this in to much more intuitive logic.

soligen2010, 2020-01-05:
The existing code was loosing the sign, it is very wrong - good job finding this, however detecting and putting the sign back is still going to be an issue because the bits will be all wrong (for example. -2 for an int is 1111 1111 1111 1110).  You may not notice this until your tick rate exceeds your code speed so that diff becomes more negative

What I would prefer is to never loose the sign, so it doesn't have to be re-established.  I am assuming you have some hardware handy to test.  I think it can all be simplified down to 2 lines.  Can you please try this instead:

```
  delta = val % steps;
  val /= steps;

```

tablatronix, 2020-01-06:
I think I tried that originally, but I read that the behavior of modulo on signed int in C++ is implementation defined, not standard,(concerning for portable code) also it(modulo) is, I think, not very optimized. I am not very c++ literate though.

soligen2010, 2021-01-23:
I finally did a project with the right hardware to test this and put in the 2 line fix.  Seems to work.

soligen2010, 2021-01-23:
Fix is in the latest commit

closed 2021-01-23 by soligen2010