soligen2010/encoder #16: Strage Encoder behavior

mbenecke, 2019-10-14:
Hi,

i have a strange behavior with my Rotary Encoders. I used two different types, to be sure. But both show the same:

Setup:
- Arduino Uno  (China Clone, old Firmware)
- a Encoder (zuczug Rotary Encoder Module for Arduino Ali***) or a KY-040 from a Arduino starter Kit. Similar Effect
- Dupont Connector

I can rotate left or right, it has the same effect:
The Logic Analyser looks ok, or at least sufficient. 
[image]
[image]

But the output of the Serial Monitor is
```
Mon Oct 14 13:49:05 (master)> platformio.exe device monitor                              
Miniterm on COM4  115200,8,N,1                                            
Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H                    
Encoder Value: 0                                                                         
Encoder Value: -1                                                                        
Encoder Value: 0                                                                         
Encoder Value: -1                                                                        
Encoder Value: 0                                    
Encoder Value: -1                                                                        
Encoder Value: 0                                                                         
Encoder Value: -1                                                                        
Encoder Value: 0                                                                         
Encoder Value: -1                                                                        
Encoder Value: 0                                                                         
Encoder Value: -1
```

and the Code of the Arduino is your example:
using ENC_HALFSTEP or ENC_DECODER makes no difference.

```
#include <ClickEncoder.h>
#include <TimerOne.h>

ClickEncoder *encoder;
int16_t last, value;
#define ENCODER_PIN_A    5
#define ENCODER_PIN_B    6
#define WITHOUT_BUTTON 1
/* #define ENCODER_PUSH_BUTTON A3 */
#define ENC_HALFSTEP 1

void timerIsr() {
  encoder->service();
}

void setup() {
  Serial.begin(115200);
  /* encoder = new ClickEncoder(ENCODER_PIN_A, ENCODER_PIN_B, ENCODER_PUSH_BUTTON,1); */
  encoder = new ClickEncoder(ENCODER_PIN_A, ENCODER_PIN_B,-1);

  Timer1.initialize(1000);
  Timer1.attachInterrupt(timerIsr); 

  last = -1;
}

void loop() {  
  value += encoder->getValue();

  if (value != last) {
    last = value;
    Serial.print("Encoder Value: ");
    Serial.println(value);
  }

  ClickEncoder::Button b = encoder->getButton();
  if (b != ClickEncoder::Open) {
    Serial.print("Button: ");
    #define VERBOSECASE(label) case label: Serial.println(#label); break;
    switch (b) {
      VERBOSECASE(ClickEncoder::Pressed);
      VERBOSECASE(ClickEncoder::Held)
      VERBOSECASE(ClickEncoder::Released)
      VERBOSECASE(ClickEncoder::Clicked)
      case ClickEncoder::DoubleClicked:
          Serial.println("ClickEncoder::DoubleClicked");
          encoder->setAccelerationEnabled(!encoder->getAccelerationEnabled());
          Serial.print("  Acceleration is ");
          Serial.println((encoder->getAccelerationEnabled()) ? "enabled" : "disabled");
        break;
    }
  }    
}
```

Build Process looks OK, here the Output to verify the Version Numbers. (Windows Workstation)
```
Mon Oct 14 13:48:19 (master)> platformio run -e uno --target upload
Processing uno (platform: atmelavr; board: nanoatmega328; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/nanoatmega328.html
PLATFORM: Atmel AVR 1.15.0 > Arduino Nano ATmega328
HARDWARE: ATMEGA328P 16MHz, 2KB RAM, 30KB Flash
PACKAGES: toolchain-atmelavr 1.50400.190710 (5.4.0), framework-arduinoavr 4.1.2, tool-avrdude 1.60300.190424 (6.3.0)
Converting encoder.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 11 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ClickEncoder>
|-- <TimerOne>
Compiling .pio\build\uno\src\encoder.ino.cpp.o
C:/Users/Marco/work/TonUINO/src/encoder.ino: In function 'void loop()':
C:/Users/Marco/work/TonUINO/src/encoder.ino:40:12: warning: enumeration value 'Open' not handled in switch [-Wswitch]
     switch (b) {
            ^
C:/Users/Marco/work/TonUINO/src/encoder.ino:40:12: warning: enumeration value 'Closed' not handled in switch [-Wswitch]
Linking .pio\build\uno\firmware.elf
Checking size .pio\build\uno\firmware.elf
Building .pio\build\uno\firmware.hex
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [==        ]  18.6% (used 380 bytes from 2048 bytes)
PROGRAM: [=         ]  13.6% (used 4174 bytes from 30720 bytes)
Configuring upload protocol...
AVAILABLE: arduino
CURRENT: upload_protocol = arduino
Looking for upload port...
Use manually specified: COM4
Uploading .pio\build\uno\firmware.hex

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: reading input file ".pio\build\uno\firmware.hex"
avrdude: writing flash (4174 bytes):

Writing | ################################################## | 100% 1.19s

avrdude: 4174 bytes of flash written
avrdude: verifying flash memory against .pio\build\uno\firmware.hex:
avrdude: load data flash data from input file .pio\build\uno\firmware.hex:
avrdude: input file .pio\build\uno\firmware.hex contains 4174 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.91s

avrdude: verifying ...
avrdude: 4174 bytes of flash verified

avrdude: safemode: Fuses OK (E:00, H:00, L:00)

avrdude done.  Thank you.

=============================================================== [SUCCESS] Took 8.94 seconds ===============================================================
```

If you need more Information, please do not hesitate to drop me a mail.
I've run out of ideas of troubleshooting.

I will appreciate your help with this situation.
Cheers, bye
Marco

yet-another-average-joe, 2019-10-14:
I had a similar behaviour. Defining steps per notch as 4 solved the problem.

`encoder = new ClickEncoder(ENCODER_PIN_A, ENCODER_PIN_B, ENCODER_PIN_BUTTON, 4);`

Give it a try... Play with steps pet notch. It could solve your problem. It solved mine, with this very common encoder. Just another thing : I had many problems with hardware debouncing (RC filters + Shmitt trigger). It completely messed up the timings ; this library does not need hardware debouncing (at least for me...)

mbenecke, 2019-10-15:
Hi,

thanks for your suggestion, i checked it:
`  encoder = new ClickEncoder(ENCODER_PIN_A, ENCODER_PIN_B,-1,1);
`
one turn right and one turn left:
```
Encoder Value: 0
Encoder Value: -1
Encoder Value: 0
Encoder Value: -1
Encoder Value: 0
```
also:
`  encoder = new ClickEncoder(ENCODER_PIN_A, ENCODER_PIN_B,-1,2);
`
one turn right and one turn left:
```
Encoder Value: 0
Encoder Value: -1
Encoder Value: 0
Encoder Value: -1
Encoder Value: 0
```
and:
`  encoder = new ClickEncoder(ENCODER_PIN_A, ENCODER_PIN_B,-1,4);
`
one turn right and one turn left:
```
Encoder Value: 0
Encoder Value: -1
Encoder Value: 0
Encoder Value: -1
Encoder Value: 0
```

ok, lets rotate the other settings too:
```
...
#define ENC_DECODER (1 << 2)
void setup() {
  encoder = new ClickEncoder(ENCODER_PIN_A, ENCODER_PIN_B, ENCODER_PUSH_BUTTON,1);
...
```
one turn right and one turn left:
```
Encoder Value: 0
Encoder Value: -1
Encoder Value: 0
Encoder Value: -1
Encoder Value: 0
```
and:
```
...
#define ENC_DECODER (1 << 2)
void setup() {
  encoder = new ClickEncoder(ENCODER_PIN_A, ENCODER_PIN_B, ENCODER_PUSH_BUTTON,2);
...
```
one turn right and one turn left:
```
Encoder Value: 0
Encoder Value: -1
Encoder Value: 0
Encoder Value: -1
Encoder Value: 0
```
next:
```
...
#define ENC_DECODER (1 << 2)
void setup() {
  encoder = new ClickEncoder(ENCODER_PIN_A, ENCODER_PIN_B, ENCODER_PUSH_BUTTON,4);
...
```
one turn right and one turn left (nothing happens):
```
Encoder Value: 0
```
ok, lets try ENC_HALFSTEP
```
...
#define ENC_HALFSTEP 1
void setup() {
  encoder = new ClickEncoder(ENCODER_PIN_A, ENCODER_PIN_B, ENCODER_PUSH_BUTTON,1);
...
```
one turn right and one turn left:
```
Encoder Value: 0
Encoder Value: -1
Encoder Value: 0
Encoder Value: -1
Encoder Value: 0
```
and:
```
...
#define ENC_HALFSTEP 1
void setup() {
  encoder = new ClickEncoder(ENCODER_PIN_A, ENCODER_PIN_B, ENCODER_PUSH_BUTTON,2);
...
```
one turn right and one turn left:
```
Encoder Value: 0
Encoder Value: -1
Encoder Value: 0
Encoder Value: -1
Encoder Value: 0
```
and last but not least:
```
...
#define ENC_HALFSTEP 1
void setup() {
  encoder = new ClickEncoder(ENCODER_PIN_A, ENCODER_PIN_B, ENCODER_PUSH_BUTTON,4);
...
```
one turn right and one turn left:
```
Encoder Value: 0
Encoder Value: -1
Encoder Value: 0
Encoder Value: -1
Encoder Value: 0
```

yet-another-average-joe, 2019-10-15:
I tested the code from your 1st post on a Nano clone, with A to D5, B to D6, and "CLK" to GND ; no hardware deboucing (no external pullups no capacitors no Schmitt triggers). Encoder library Master branch, TimerOne by Paul Stoffregen ; default switches in the header file. Works flawlessly.

```
Encoder Value: 0
Encoder Value: -1
Encoder Value: -2
Encoder Value: -3
Encoder Value: -4
Encoder Value: -5
Encoder Value: -4
Encoder Value: -3
Encoder Value: -2
Encoder Value: -1
Encoder Value: 0
Encoder Value: 1
Encoder Value: 2
Encoder Value: 3
Encoder Value: 4
Encoder Value: 5
```
Couldn't reproduce your problem, even with wrong wiring (just does not work at all, as expected).

(about the 4 steps per notch : i remember now... I had to set it to this value for the STM32, or it gets too many events due to bouncing and the 72MHz clock)

mbenecke, 2019-10-16:
Thanks, i bought a set of five KY-040 Encoders and an other Arduino Uno. Maybe the new ones work together.
I will come back with my results, thanks for your support.

yet-another-average-joe, 2019-10-17:
I reproduced the exact behaviour you describe !!!

0 -1  0 -1 and so on

I just swapped the A or B pin with the switch pin !!! (forgot to try this one yesterday...)

I was interrested in your problem because I remember I had this issue a while ago. But I didn't remember how I solved it.

Your problem is just a wiring error.

mbenecke, 2019-10-19:
After switching everything (Encoder, Connection, Uno) the Library works like a charm.
I will now switch back to the original setup, one by one, and then i will check the setup with an continue checker. hope i'm able detect the root cause.

But this is no need to continue here, as mentioned in the beginning, it is definitiv not an issue with the ClickEncoder.h!

Thanks for your help

closed 2019-10-19 by mbenecke