soligen2010/encoder #18: Erratic reading of 2nd encoder in separate functions

tcarney57, 2021-05-31:
I'm using this fork of ClickEncoder to read from two encoders/buttons. After extensive tests, I'm still having trouble reading clicks of the 2nd encoder (I'm calling it "2nd" because it is the second instantiation). A test program shows I can get accurate and expected results (encoders register 1 increment for each click, and button clicks--double-clicked, held, released, etc.--return their expected values) when both encoders are handled in the loop() function. 

In my actual application, the 1st encoder/button works perfectly as expected (in loop()), and the _button_ of the 2nd encoder/button also works as expected--to call another function which I will use for a menu. Once inside that menu function, however, I cannot get the 2nd rotary to encode reliably. Sometimes a single indent will properly be registered, sometimes not. If it's turned several indents at a time, it will sometimes register, but only if I turn fast enough. Thinking it may be the problem, I've disabled acceleration for both encoder/buttons (I don't want it anyway). That has not solved the problem. By the way, the button of the 2nd encoder/button does work as expected to exit the menu function with a press-and-release (that is, the menu function closes when it receives the "release" code (4) is returned. 

Because of the erratic behavior, I'm guessing there may be an issue with the timer-based interrupt, but I'm not knowledgeable enough in the library to know--and certainly not enough to fix it. Is anyone else aware of 1) problems related to use of ClickEncoder in separate functions, and 2) what can be done about them?

Thanks!

soligen2010, 2021-06-01:
I am just guessing, but maybe with multiple encoders the ISR is taking too long.  I have done multiple buttons with no issues, but have never done multiple encoders.  A couple things you can try:

Try running without using a timer ISR (see the ESP8266 example) - basically you just put the call to service in Loop.

In the .h try changing the value of ENC_DECODER   to ENC_FLAKEY.

soligen2010, 2021-12-28:
Any update on this? Or, can thi sbe closed?

soligen2010, 2023-09-08:
closing due to inactivity

closed 2023-09-08 by soligen2010