soligen2010/encoder #19: This is not an issue, but a dumb question

fyrus7, 2021-09-11:
Can a detent detect as a click?

marmilicious, 2021-09-11:
Do you mean a button click?  
A detent should not be detected as a button click.

Can you provide a bit more info?

tcarney57, 2021-09-11:
> Can a detent detect as a click?

Are trying to use a detent this way, or is this happening and you don't know why? Programmatically, a "click" and a "detent" can be used in the same way. Both are events which you can detect and act accordingly. With many encoders--especially inexpensive ones--they're electrically similar as well. Both use make-and-break contacts, though some encoders are optical while their associated "buttons" use contacts. If you're getting a "click" when you're expecting a "detent," then you have the device wired wrong, or maybe the pins you're using need pull-ups which can also result in unexpected results. --Todd

soligen2010, 2021-09-11:
detents are for the encoder when you turn the knob.  button clicks are for the button function.  Many (but not all) encoders have a momentary button built into them that is activated when you press the shaft in.  These are two functions are completely independent.

closed 2021-12-28 by soligen2010