soligen2010/encoder #21: Not working nicely with VSCode/Platformio

snowrodeo, 2021-12-09:
Some kind of reorder error is showing up with building with Platformio.

`src/main.cpp:57:12: warning: enumeration value 'Closed' not handled in switch [-Wswitch]
In file included from .pio/libdeps/featheresp32/ClickEncoder/ClickEncoder.cpp:12:0:
.pio/libdeps/featheresp32/ClickEncoder/ClickEncoder.h: In constructor 'ClickEncoder::ClickEncoder(int8_t, int8_t, int8_t, uint8_t, bool)':
.pio/libdeps/featheresp32/ClickEncoder/ClickEncoder.h:165:8: error: 'ClickEncoder::buttonHeldEnabled' will be initialized after [-Werror=reorder]
   bool buttonHeldEnabled;
        ^
.pio/libdeps/featheresp32/ClickEncoder/ClickEncoder.h:158:8: error:   'bool ClickEncoder::accelerationEnabled' [-Werror=reorder]
   bool accelerationEnabled;
        ^
.pio/libdeps/featheresp32/ClickEncoder/ClickEncoder.cpp:44:1: error:   when initialized here [-Werror=reorder]
 ClickEncoder::ClickEncoder(int8_t A, int8_t B, int8_t BTN, uint8_t stepsPerNotch, bool active)
 ^
In file included from .pio/libdeps/featheresp32/ClickEncoder/ClickEncoder.cpp:12:0:
.pio/libdeps/featheresp32/ClickEncoder/ClickEncoder.h:158:8: error: 'ClickEncoder::accelerationEnabled' will be initialized after [-Werror=reorder]
   bool accelerationEnabled;
        ^
.pio/libdeps/featheresp32/ClickEncoder/ClickEncoder.h:154:20: error:   'volatile int16_t ClickEncoder::delta' [-Werror=reorder]
   volatile int16_t delta;
                    ^
.pio/libdeps/featheresp32/ClickEncoder/ClickEncoder.cpp:44:1: error:   when initialized here [-Werror=reorder]
 ClickEncoder::ClickEncoder(int8_t A, int8_t B, int8_t BTN, uint8_t stepsPerNotch, bool active)
 ^
In file included from .pio/libdeps/featheresp32/ClickEncoder/ClickEncoder.cpp:12:0:
.pio/libdeps/featheresp32/ClickEncoder/ClickEncoder.h:163:19: error: 'ClickEncoder::button' will be initialized after [-Werror=reorder]
   volatile Button button;
                   ^
.pio/libdeps/featheresp32/ClickEncoder/ClickEncoder.h:156:20: error:   'volatile uint8_t ClickEncoder::steps' [-Werror=reorder]
   volatile uint8_t steps;
                    ^
.pio/libdeps/featheresp32/ClickEncoder/ClickEncoder.cpp:44:1: error:   when initialized here [-Werror=reorder]
 ClickEncoder::ClickEncoder(int8_t A, int8_t B, int8_t BTN, uint8_t stepsPerNotch, bool active)
 ^
In file included from .pio/libdeps/featheresp32/ClickEncoder/ClickEncoder.cpp:12:0:
.pio/libdeps/featheresp32/ClickEncoder/ClickEncoder.h:156:20: error: 'ClickEncoder::steps' will be initialized after [-Werror=reorder]
   volatile uint8_t steps;
                    ^
.pio/libdeps/featheresp32/ClickEncoder/ClickEncoder.h:150:10: error:   'int8_t ClickEncoder::pinA' [-Werror=reorder]
   int8_t pinA;
          ^
.pio/libdeps/featheresp32/ClickEncoder/ClickEncoder.cpp:44:1: error:   when initialized here [-Werror=reorder]
 ClickEncoder::ClickEncoder(int8_t A, int8_t B, int8_t BTN, uint8_t stepsPerNotch, bool active)
 ^
In file included from .pio/libdeps/featheresp32/ClickEncoder/ClickEncoder.cpp:12:0:
.pio/libdeps/featheresp32/ClickEncoder/ClickEncoder.h: In constructor 'ClickEncoder::ClickEncoder(int8_t, bool)':
.pio/libdeps/featheresp32/ClickEncoder/ClickEncoder.h:165:8: error: 'ClickEncoder::buttonHeldEnabled' will be initialized after [-Werror=reorder]
   bool buttonHeldEnabled;
        ^
.pio/libdeps/featheresp32/ClickEncoder/ClickEncoder.h:158:8: error:   'bool ClickEncoder::accelerationEnabled' [-Werror=reorder]
   bool accelerationEnabled;
        ^
.pio/libdeps/featheresp32/ClickEncoder/ClickEncoder.cpp:74:1: error:   when initialized here [-Werror=reorder]
 ClickEncoder::ClickEncoder(int8_t BTN, bool active)
 ^
In file included from .pio/libdeps/featheresp32/ClickEncoder/ClickEncoder.cpp:12:0:
.pio/libdeps/featheresp32/ClickEncoder/ClickEncoder.h:158:8: error: 'ClickEncoder::accelerationEnabled' will be initialized after [-Werror=reorder]
   bool accelerationEnabled;
        ^
.pio/libdeps/featheresp32/ClickEncoder/ClickEncoder.h:154:20: error:   'volatile int16_t ClickEncoder::delta' [-Werror=reorder]
   volatile int16_t delta;
                    ^
.pio/libdeps/featheresp32/ClickEncoder/ClickEncoder.cpp:74:1: error:   when initialized here [-Werror=reorder]
 ClickEncoder::ClickEncoder(int8_t BTN, bool active)
 ^
In file included from .pio/libdeps/featheresp32/ClickEncoder/ClickEncoder.cpp:12:0:
.pio/libdeps/featheresp32/ClickEncoder/ClickEncoder.h:163:19: error: 'ClickEncoder::button' will be initialized after [-Werror=reorder]
   volatile Button button;
                   ^
.pio/libdeps/featheresp32/ClickEncoder/ClickEncoder.h:156:20: error:   'volatile uint8_t ClickEncoder::steps' [-Werror=reorder]
   volatile uint8_t steps;
                    ^
.pio/libdeps/featheresp32/ClickEncoder/ClickEncoder.cpp:74:1: error:   when initialized here [-Werror=reorder]
 ClickEncoder::ClickEncoder(int8_t BTN, bool active)
 ^
In file included from .pio/libdeps/featheresp32/ClickEncoder/ClickEncoder.cpp:12:0:
.pio/libdeps/featheresp32/ClickEncoder/ClickEncoder.h:167:22: error: 'ClickEncoder::analogInput' will be initialized after [-Werror=reorder]
   bool analogInput = false;
                      ^
.pio/libdeps/featheresp32/ClickEncoder/ClickEncoder.h:150:10: error:   'int8_t ClickEncoder::pinA' [-Werror=reorder]
   int8_t pinA;
          ^
.pio/libdeps/featheresp32/ClickEncoder/ClickEncoder.cpp:74:1: error:   when initialized here [-Werror=reorder]
 ClickEncoder::ClickEncoder(int8_t BTN, bool active)
 ^
cc1plus: some warnings being treated as errors
*** [.pio/build/featheresp32/lib82b/ClickEncoder/ClickEncoder.cpp.o] Error 1`

soligen2010, 2021-12-09:
I don't use Platform IO or VSCode, so I cant help with this.  I will leave this issue open for a while in case someone else can help

tablatronix, 2021-12-12:
Same issue, no idea, I tried disabling this error, i tried reordering, must have something to do with the directives

soligen2010, 2021-12-28:
Any resolution here?

soligen2010, 2023-09-08:
No activity, so closing

closed 2023-09-08 by soligen2010